jwise/HoRNDIS #150: Couldnt make it in Moterrey 12.1

larts85, 2021-12-19:
[image]

give me an error message and transfer to the link bellow.

https://support.apple.com/en-us/HT210999

chris1111, 2021-12-19:
https://github.com/jwise/HoRNDIS/issues/148

chris1111, 2022-01-22:
> It didn't install on a clean and fully updated Big Sur, one of the instructions did not work :(

[
](https://github.com/jwise/HoRNDIS/issues/151#issuecomment-1019189944)

duckfly-tw, 2022-03-14:
on Moterrey 12 it may be OK, BUT on 12.1 I just can't make it work when using chris1111's port, even following all the suggested steps. System show messages said that it prevent old way kext be loaded, and developers need to use new way to write their kernel extension.