jwise/HoRNDIS #151: I know Joshua no longer develops, but Chris1111 has archived his port already

zaphodd, 2022-01-19:
I know Joshua no longer develops, but Chris1111 has archived his port already, so i can not tell Chris of the issue i have.
so i ask the only Horndis community i am able to.
the instructions:
Make sur your Monterey or Big Sur is on SIP Enable
Double clic Consent-HoRNDIS.command
Boot in Recovery open Utility / Terminal
Type: the command bellow followed by Enter
csrutil disable
Type: the command bellow followed by Enter
/usr/sbin/spctl kext-consent add 54GTJ2AU36
----------------------------------------------
i get error:
Orac:~ zaphod$ /usr/sbin/spctl kext-consent add 54GTJ2AU36
spctl: failed to store new configuration.
Orac:~ zaphod$ 
 ----------------------------------------------
 but i carried on & it asked (and alowed) me to Allow the extension,
 i was happy, so i rebooted, did 'csrutil enable' again - now it is non working again.
 i am on Big Sur clean install as of today and full updates.
 I  was so happy someone updated Horndis, but sad when i saw the github was archived after only 23 days
 so i could not ask for help. only 1 person opened an issue before the project was marked as read only 23 days in :)
 i am feeling sad now.

chris1111, 2022-01-22:
Follow exactly the guide its work
SIP must be enable before clicking **Consent-HoRNDIS.command**
[GUIDE Consent HoRNDIS.pdf](https://github.com/jwise/HoRNDIS/files/7918185/GUIDE.Consent.HoRNDIS.pdf)

zaphodd, 2022-02-02:
Chris, i did exactly that, in those instructions letter for letter, precisely, but the error message is still the same.
it failed to store the new configuration, as it told me before.
i tried via sudo, but it still failed. I had SIP disabled, you say to do it whilst SIP is enabled, so i will install Big Sur again now
and after the install and updates are done, the first thing i will do is this. I will write back after i do this with SIP Enabled as you instruct.

phu54321, 2022-04-04:
Chris, could you compile a notarized build for arm64? For iPhone app execution SIP should be enabled, so we need arm64 build. Thanks.

AlfredJKwack, 2022-05-14:
Indeed a notarized M1 (arm64**e**) buid would be quite nice.