jwise/HoRNDIS #153: Support for Monterey 12.1 M1 Chip

SoumadeepChoudhury, 2022-01-30:
## PLZ build the HoRNDIS in such a way that it natively supports M1 chip and Monterey 12 or later. 

### So that we do not need to disable SIP for installation.....

misinglink, 2022-02-17:
yes please

AlekssZeltins, 2022-03-16:
I would Like that also

MrCoding-cl, 2022-04-07:
pls

maurocos, 2022-04-20:
pls

AlfredJKwack, 2022-05-14:
Have you tried PR #149? Not sure it covers M1 though it does cover Monterey.

altitudedashboard, 2022-09-04:
Pleaseeee

AlfredJKwack, 2022-09-05:
@altitudedashboard see my comment above. It gets you there.

altitudedashboard, 2022-09-06:
This workaround disables SIP which in turn on M1 chips disables the ability to run iOS apps. For some people this is not an option.

inspiredearth, 2022-09-27:
Agreed. It would be great to have a correctly signed version of the kext which will run on M1.

samip5, 2022-10-16:
> This workaround disables SIP which in turn on M1 chips disables the ability to run iOS apps. For some people this is not an option.

I don't think one can run iOS apps anymore on M1 macs as Apple has disabled it in MacOS 12.1 or is that just the ones that are not meant to be run on it by devs?