jwise/HoRNDIS #156: Mac OS 12

edludlowedit, 2022-05-20:
Anyone had any luck getting this going Mac OS 12 Monterey?

Barathwaja, 2022-06-21:
Yes it's not working for me as well. I'm using Mac OS 12.4 Monterey

ea7kir, 2022-07-09:
Will there be any further development or is this project dead?

tfvglin, 2022-07-10:
is there anyone working on this project?

MicahZheng, 2022-07-19:
https://youtu.be/32lM27TGNFM
this video worked well for me ,I'm using Mac OS 12.4 Monterey (Intel Chip)

First you need to disable SIP in recovery mode using "csrutil disable"

[HoRNDIS.kext.zip](https://github.com/jwise/HoRNDIS/files/9137322/HoRNDIS.kext.zip)

@ea7kir @tfvglin @Barathwaja @

altitudedashboard, 2022-07-21:
Any alternative to this which doesn't require disabling SIP?

egigoka, 2022-08-21:
> https://youtu.be/32lM27TGNFM this video worked well for me ,I'm using Mac OS 12.4 Monterey (Intel Chip)
> 
> First you need to disable SIP in recovery mode using "csrutil disable"
> 
> [HoRNDIS.kext.zip](https://github.com/jwise/HoRNDIS/files/9137322/HoRNDIS.kext.zip)
> 
> @ea7kir @tfvglin @Barathwaja @

csrutil enable --kext 
is enough btw

LizGalanis, 2022-09-08:
I'm trying to install this on Monterey 12.5.1 M1 chip with SIP disabled but I get an error saying it needs to be updated by the developer. Anyone managed to use this recently?

egigoka, 2022-09-08:
> I'm trying to install this on Monterey 12.5.1 M1 chip with SIP disabled but I get an error saying it needs to be updated by the developer. Anyone managed to use this recently? 

Did you tried builds specially for arm? https://github.com/nEdAy/HoRNDIS-for-M1 like this one?

timredfern, 2022-09-15:
On OSX Monterey 12.5.1 (intel chip), I disabled SIP with csrutil and used the download package to install. It seems to work well, impressive performance.

LizGalanis, 2022-09-23:
> https://github.com/nEdAy/HoRNDIS-for-M1

Thanks, I have tried this one but still not working. My computer now says it is successful when I install the app, however trying to tether my phone does not work. I have discovered that in the system information it has listed the Horndis app under 'disabled software' for no user consent. I don't know how to consent! Can anyone help with this?

LizGalanis, 2022-09-23:
Is anyone able (and willing) to get on a call with me to help me make this work?

egigoka, 2022-09-23:
> Is anyone able (and willing) to get on a call with me to help me make this work? 

https://t.me/filthymold
Contact me, I'll try to help