jwise/HoRNDIS #163: Unable to load 9.2 HoRNDIS on MAC Ventura (M2) 13.2.1 (22D68)

tbonemain, 2023-03-07:
This in needed for DJI Assistant 2 install. https://www.dji.com/downloads/softwares/assistant-dji-2
I get this error when its loading HoRNDIS:

This package is incompatible with this version of macOS. The package is trying to install content to the system volume. Contact the software manufacturer for assistance.

I tried getting latest 9.2 version here:
https://www.joshuawise.com/horndis#available_versions

And get a different error: The installation failed. The installer encountered an error that cause the install to fail.  Please contact the software manufacturer for assistance.

MacBook Air 2022
OSX 13.2.1 (22D68)
Apple M2

raziele-dev, 2023-03-07:
Please help having the same issue. Thank you very much.

budgetdevv, 2023-04-16:
Likewise

Traace, 2023-05-22:
I don't think users are still allowed to load 3rd party kext drivers with MacOS Ventura. It got replaced with DriverKit extensions (dext). This project would need a rewrite to support dext, but I doubt that will happen.

Ask DJI to deliver a software update, their last update is from 2018. Before MacOS Ventura or ARM Macs even existed...

coryking, 2023-06-19:
For what it's worth I have a PlutoSDR and the first thing it says in order to use my device is to install this driver.

maximveksler, 2023-08-27:
> This in needed for DJI Assistant 2 install. https://www.dji.com/downloads/softwares/assistant-dji-2 I get this error when its loading HoRNDIS:
>
> This package is incompatible with this version of [...]

I would strongly suggest contacting DJI and have them support this open source project.

d235j, 2023-08-27:
As a former maintainer I have considered updating the README to indicate that this driver is obsolete and unsupported. It will never work reliably on newer macOS versions, and is likely to increase system instability in general.

The correct fix is for device vendors to update their firmware to support USB CDC-ACM and/or ECM, which is the standardized mechanism for USB-Ethernet and does not require a driver on macOS. This mechanism works on newer versions of Windows, but RNDIS can be enabled alongside on the device/firmware side for older versions.

Support for this mechanism is present in the Linux kernel and merely requires appropriate configuration. Newer BeagleBone images have this. However, it does not seem that PlutoSDR does, so it would be worth contacting them, and DJI firmware also does not. I’ve tried to contact DJI multiple times to get them to make this change but they don’t seem interested.

d235j, 2023-08-27:
As a former maintainer I think it would be worth updating the README to indicate that this driver is obsolete and unsupported. It will never work reliably on newer macOS versions, and is likely to increase system instability in general.

The correct fix is for device vendors to update their firmware to support USB CDC-ACM and/or ECM, which is the standardized mechanism for USB-Ethernet and does not require a driver on macOS. This mechanism works on newer versions of Windows, but RNDIS can be enabled alongside on the device/firmware side for older versions.

Support for this mechanism is present in the Linux kernel and merely requires appropriate configuration. Newer BeagleBone images have this. However, it does not seem that PlutoSDR does, so it would be worth contacting them, and DJI firmware also does not. I’ve tried to contact DJI multiple times to get them to make this change but they don’t seem interested.

jwise, 2023-08-27:
I also would suggest contacting DJI in the same way.

Actually, DJI is basically why I do not maintain this project anymore.  I recently even bought myself a Mac Studio but I don't really plan to do anything with forward porting this.

I have often fantasized about setting up an autoresponder to people who mail me about help getting HoRNDIS to work with their DJI products on modern Macs, telling them that DJI has profited off of my work and never sent me so much as a thank you, let alone offering to financially support the development of it, and that if they need support with their DJI products, they should call DJI.  DJI is probably lucky that I just can't be bothered.  Life is too long to pick fights like that.

ksearcher, 2025-02-13:
Any idea how we can connect android to Mac for usb tethering ...???, as HoRNDIS req for Mac driver to support that, I am unable to do that with my M1 Mac.