jwise/HoRNDIS #170: WARNING to Users: Don't Use This Code --- NOTE to the Developer: Sign Your Kernel Extension! - It's Not That Difficult

JeffHochberg, 2023-11-13:
I would suggest that anyone considering installing this not do it. The developer needs to do what's right for the user. Installing software should NEVER require booting to Recovery Mode and running csrutil.

Why don't you make life easier for the users that want to use your code? Sign your kernel extension so it doesn't require dialing back the security built into the operating system for a reason.

The only reason for not doing it would be not wanting to spend the $99 USD to join the Apple Developer's program so you can get a signing certificate.

https://developer.apple.com/library/archive/documentation/Security/Conceptual/System_Integrity_Protection_Guide/KernelExtensions/KernelExtensions.html#:~:text=You%20should%20sign%20a%20kernel,filling%20out%20the%20required%20details.

Also - I would strongly encourage you to test and certify your kernel extension(s) against modern versions of macOS (including Ventura and Sonoma) to ensure that it works as advertised on Apple Silicon.

The fact that we have to put our Macs through these motions increases the potential for instability and the need to reinstall macOS later. I started down the process of installing it, then decided it was just not worth it. In fact, when you run:

```
$ csrutil enable --without kext
csrutil: requesting an unsupported configuration. This is likely to break in the future and leave your machine in an unknown state.
```

I can't imagine why anyone would continue past that point. There's no reason to - so long as the developer does their due diligence - which is not what's happening here.

I'll use my Windows computer to connect to my Android device(s) instead. At least in that case, I don't need to disable driver signing to use it.

d235j, 2023-11-13:
https://github.com/jwise/HoRNDIS/issues/163#issuecomment-1694740502

jwise, 2023-11-13:
> NOTE to the Developer: Sign Your Kernel Extension! 

Note from the developer: this is abandonware.  Don't run it.

> It's Not That Difficult

Until two months ago I did not own a mac that could sign a modern kext.  I now have a M2 Mac, but until very recently, my previous Mac was a Mid-2012 Retinabook.  The cost of the Mac Studio on my desk was dramatically more than $99.

> The developer needs to do what's right for the user. [...] Why don't you make life easier for the users that want to use your code?

Man you guys are entitled.  The developer has not told you to use this or to do anything at all with csrutil, you found this all on the internet unrelated to me.  The developer flatly refuses to "do what's right for the user" because the user worth $25 billion (DJI) has not done what's right for the developer (that is to say, so much as writing a letter of thanks).  The tragedy of the commons has struck again, go thank DJI for it.

tiagoad, 2023-11-15:
The entitlement of FOSS users never ceases to amaze me. Thanks for your work, and you gave a perfect response. Cheers

JeffHochberg, 2023-11-24:
Because of the wonders of FOSS! I have you to do it for me for $0.00!

closed 2023-11-24 by JeffHochberg

whitequark, 2024-02-26:
Wow, what a complete cunt.