jwise/HoRNDIS #171: No working in MacOS 15.0 Sequoia

LiveRock, 2024-10-04:
No working in MacOS 15.0 Sequoia
See error:
[image]

Aaronchannn, 2024-10-24:
same same

Duarte475, 2024-10-30:
Also the same issue, the installer installs the kext on /Library/Extensions but not in /System/Library/Extensions and it doesn't work

ivanushka123amongus, 2024-12-05:
You can try kext there https://github.com/TomHeaven/HoRNDIS
It’s fork for macOS 12+ that is working on Sequoia. I tried it on hackitnosh bu putting it into kernel/extensions and it worked perfectly

p0lARbEAR, 2024-12-20:
> You can try kext there https://github.com/TomHeaven/HoRNDIS It’s fork for macOS 12+ that is working on Sequoia. I tried it on hackitnosh bu putting it into kernel/extensions and it worked perfectly

I tried TomHeaven's version on my M3 14.3 Sonoma. I did the following in recovery mode

1. disables the kext signing requirement ("csrutil enable --without kext")

2. Allow user management of kernel extensions from identified developers
    Allow remote management of kernel extensions and automatic software updates

then it works like a charm

camerakang, 2025-03-24:
可以尝试安装https://github.com/jwise/HoRNDIS/files/3719733/HoRNDIS-9.2-catalina_install.pkg.zip
他解决了我的问题，