jwise/HoRNDIS #176: How to use on macOS Tahoe 26.0?

34alp, 2025-09-25:
How to use on macOS Tahoe 26.0?

Toshasun, 2025-09-27:
Прекрасно работает на MacOS Tahoe 26.0
👇👇👇
https://github.com/iKeramat/HoRNDIS/releases/latest

AA-EION, 2026-09-25:
If you're still stuck on this: I've been working on TetherKitNext, a free, open-source app that brings Android USB tethering (RNDIS) back to modern Macs.

Unlike the older options, there's nothing to set up:

no kernel extension, so no turning off SIP or lowering security settings;
no compiling, no Homebrew, no Terminal commands.

It's a signed and notarized app for Apple Silicon and Intel (macOS 14+). You download the DMG, drag it to Applications, click Connect, and you're online.

https://github.com/AA-EION/TetherKitNext

If it doesn't work with your phone, please open an issue. Feedback really helps.

(Disclosure: I'm part of the team behind it, at Issen Software Group. It's a fork of XiaoMiku01's TetherKit.)

sujitagarwal, 2026-10-30:
If you're on Apple Silicon / macOS 26: HoRNDIS is a kernel extension and won't load there, so it can't be fixed by reinstalling. I wrote a userspace alternative that doesn't need a kext, DriverKit or SIP changes: https://github.com/sujitagarwal/rehorndis

One-line install, and a menu bar switch to turn tethering on/off. I've tested it on macOS 26 (Apple Silicon) with a Samsung phone only, so reports from other phones are welcome as issues on that repo. It's IPv4-only and slower than a native driver, since it's Python.

(Disclosure: I'm the author.)