jwise/HoRNDIS #177: Not working in MacOS Tahoe 26.0

Joker781998, 2025-09-25:
https://www.reddit.com/media?url=https%3A%2F%2Fpreview.redd.it%2Fhorndis-9-2-not-working-in-macos-tahoe-26-0-v0-orfuyizsi3rf1.png%3Fwidth%3D627%26format%3Dpng%26auto%3Dwebp%26s%3D3b6b863ea95e9c9eb0de25cb2b4a76fc9231e07d

Toshasun, 2025-09-27:
Прекрасно работает на MacOS Tahoe 26.0
👇👇👇
https://github.com/iKeramat/HoRNDIS/releases/latest

Richiardi72, 2025-10-01:
Not with me.

Toshasun, 2025-10-01:
> Not with me.

https://github.com/iKeramat/HoRNDIS?tab=readme-ov-file#installation-preparing

Richiardi72, 2025-10-02:
@Toshasun I get on a loop where it never gets approved.
[image]
[image]
[image]

Toshasun, 2025-10-02:
Here is the video I was installing
https://www.youtube.com/watch?v=zlhc1V01EK4&list=LL&index=379

wansho, 2026-04-15:
> Прекрасно работает на MacOS Tahoe 26.0 👇👇👇 https://github.com/iKeramat/HoRNDIS/releases/latest

works for me, thx

[image]

[image]

AA-EION, 2026-09-25:
If you're still stuck on this: I've been working on TetherKitNext, a free, open-source app that brings Android USB tethering (RNDIS) back to modern Macs.

Unlike the older options, there's nothing to set up:

no kernel extension, so no turning off SIP or lowering security settings;
no compiling, no Homebrew, no Terminal commands.

It's a signed and notarized app for Apple Silicon and Intel (macOS 14+). You download the DMG, drag it to Applications, click Connect, and you're online.

https://github.com/AA-EION/TetherKitNext

If it doesn't work with your phone, please open an issue. Feedback really helps.

(Disclosure: I'm part of the team behind it, at Issen Software Group. It's a fork of XiaoMiku01's TetherKit, i found out @prostec-labs had the same name for their software as the fork i based the work on, so be aware that I'm not related to none of those.)

sujitagarwal, 2026-10-30:
If you're on Apple Silicon / macOS 26: HoRNDIS is a kernel extension and won't load there, so it can't be fixed by reinstalling. I wrote a userspace alternative that doesn't need a kext, DriverKit or SIP changes: https://github.com/sujitagarwal/rehorndis

One-line install, and a menu bar switch to turn tethering on/off. I've tested it on macOS 26 (Apple Silicon) with a Samsung phone only, so reports from other phones are welcome as issues on that repo. It's IPv4-only and slower than a native driver, since it's Python.

(Disclosure: I'm the author.)